DiverOfDark/KNOMI #15: Feature: scan for networks during initial setup

DiverOfDark, 2023-09-17:
It would be nice and convenient to be able to select network from scanned during initial setup instead of typing it's name manually.

Still need to leave ability to enter SSID manually because of hidden networks.

jMaddrell, 2023-09-17:
Would it be worth looking Into existing onboarding frameworks like Improv?

DiverOfDark, 2023-09-17:
I think Improv could be an interesting addition - but designing proper user experience would be really hard.

As far as I understood - user needs to have some kind of device or webpage which would use BLE to send some initial configuration datagram to Knomi with SSID and pass.

In ideal world - moonraker on raspberry pi could use improv to discover all Knomi-like displays and present a buttons in Mainsail/Fluidd to "add my display" - and then autoconfiguration would work out of the box. 
Or, we could provide some kind of shell script which would do the similar.

Next complex part - user needs to enter a bit more than wifi details. Knomi needs to know Klipper IP/hostname, new hostname for knomi, and a more coming soon - accent color, and may be some picture overrides. For all of these cases we currently use a webpage as an entrypoint.

So improv may work for us, if there was an possibility to inject iframe/custom code to mainsail/fluidd which could do the job for user - first connecting device to proper wifi, and then providing an user interface to actual configuration.
Unfortunately I think this may be really far distant future, if possible at all.

On other hand - we could try to look into libraries which provide better captive portal / wifi configuration / management - e.g. smth like https://github.com/zhouhan0126/WIFIMANAGER-ESP32 . But considering we already have custom and rich webui - I am not sure it would be that beneficial.

jMaddrell, 2023-09-18:
> I think Improv could be an interesting addition - but designing proper user experience would be really hard.
>
> As far as I understood - user needs to have some kind of device or webpage which would [...]

From my understanding once Improv has configured Wifi it will redirect the browser to the devices IP address, which would allow for the configuration of the remaining options. 

> So improv may work for us, if there was an possibility to inject iframe/custom code to mainsail/fluidd which could do the job for user - first connecting device to proper wifi, and then providing an [...]

Good point

> On other hand - we could try to look into libraries which provide better captive portal / wifi configuration / management - e.g. smth like https://github.com/zhouhan0126/WIFIMANAGER-ESP32 . But considering we already have custom and rich webui - I am not sure it would be that beneficial.
Agreed maybe it would be a nice to have for the future 👍

PR #165 (2024-02-25): Feature/choose wifi
resolves #15
closed 2024-02-25 by DiverOfDark