DiverOfDark/KNOMI #9: Automatic code formatter/linter

DiverOfDark, 2023-09-17:
Currently during PRs we often see code being reformated by IDE - that makes reading PR harder.

We need to have some kind of autoformatter which would do code format on build + committed settings for this code formatter.

jMaddrell, 2023-09-17:
I currently have format on save enabled in VS Code with clangFormat and LLVM Style. I'm happy to change this.

DiverOfDark, 2023-09-17:
Can you share any kind of readme on how to use it?
I'm coming from java/c# world and not familiar with best practices on cpp - so I'll stick to whatever in community already exists.

Also - is it possible to trigger it from CLI? I still want to use CLion trial (and beg JB to get a license for OSS project) - so ideally we'll have a pre-build script which would autoformat code both on cpp part and on svelte/ts part.

jMaddrell, 2023-09-17:
I just did brew-install clang-format and then set VSCode up like this:
[image]

I have tried CLion previously and from a quick google it also supports clang-format: https://www.jetbrains.com/help/clion/clangformat-as-alternative-formatter.html

jMaddrell, 2023-09-17:
To run from the CLI something like this would do the entire project:
```sh
clang-format src/**/*.h -i && clang-format src/**/*.cpp -I
```

jMaddrell, 2023-09-17:
We could also use https://pre-commit.com/hooks.html with the clang-format extension: https://github.com/pre-commit/mirrors-clang-format

PR #16 (2023-09-17): Implemented autoformat of sources during build
resolves #9
closed 2023-09-17 by DiverOfDark