DiverOfDark/KNOMI #22: Feature: Allow users to select their preferred method of print completion

jMaddrell, 2023-09-20:
Klipper provides several ways of calculating the completion percentage:

- file-relative (Moonraker & Fluidd default)
- file-absolute
- slicer (KNOMI default)
- filament

The user should be able to select which method is used. I have done the analysis on which endpoints are required for each method and will attach once I've had a chance to clean it up.

jMaddrell, 2023-10-30:
Requests required for each method of calculating print %

file-relative
- http://voron2.local/server/files/metadata?filename=??? && /printer/objects/query?virtual_sdcard&print_stats
-- currentPos = (req2) result.status.virtual_sdcard.file_position - (req1) result.gcode_start_byte
-- maxPos = (req1) result.gcode_end_byte - (req1) result.gcode_start_byte
-- if (currentPos > 0 && maxPos > 0) return (1 / maxPos) * currentPos

file-absolute
- /printer/objects/query?virtual_sdcard
-- result.status.virtual_sdcard.progress

slicer
- /printer/objects/query?display_status
-- result.status.display_status.progress

filament
- /printer/objects/query?print_stats & http://voron2.local/server/files/metadata?filename=???
--  result.filament_total (req2) / (req1)result.status.print_status.filament_used

jMaddrell, 2023-10-30:
One thing to note is that we can combine requests to /printer/objects/query eg
```
/printer/objects/query?display_status&virtual_sdcard&gcode_macro%20BED_MESH_CALIBRATE&gcode_macro%20G28&print_stats
```

DiverOfDark, 2024-06-01:
FWIW - as part of #247 involuntarily changed to file-absolute

jMaddrell, 2025-01-03:
i have a rough prototype for file-absolute and slicer, need to work out how to fetch file metadata with the new streaming setup

DiverOfDark, 2025-01-04:
Probably you'll have to query for the metadata one time at the start of the print and save it for current state somehow.
AFAIK file metadata is part of moonraker, not klipper api, and moonraker can't stream it.

jMaddrell, 2025-01-04:
Thanks for the tip, any thoughts on which config object to put this under? I was thinking KlipperConfig since I’ll also need the host and api key for the metadata request.

DiverOfDark, 2025-01-04:
You can do the request through websockets as well, e.g. we already subscribed to printer stats - so when print is starting you could send something like:
```
    String printerInfo = "{\n"
                         "    \"jsonrpc\": \"2.0\",\n"
                         "    \"method\": \"server.files.metadata\",\n"
                         "    \"params\": { \"filename\": \"{NEED_TO_GET_FILENAME}\" }, "
                         "    \"id\": 12345\n"
                         "}";

    esp_websocket_client_send_text(client, printerInfo.c_str(), printerInfo.length(), portMAX_DELAY);
```

and after that wait for response with if 12345
the only Q is how you can get filename - probably as part of starting

DiverOfDark, 2025-01-04:
or, extend subsription to virtual_sdcard -> to get `is_active` and `file_path` (https://www.klipper3d.org/Status_Reference.html#virtual_sdcard)

PR #415 (2025-01-04): Feature: Allow users to select their preferred method of print completion
Fixes #22
commit df65d739 (2025-01-04): Feature: Allow users to select their preferred method of print completion
Fixes #22
closed 2025-01-04 by jMaddrell