DiverOfDark/KNOMI #23: Docs: proper documentation for switching from original BTT firmware

DiverOfDark, 2023-09-23:
We need to have a proper detailed readme in repository explaining how to install firmware from this repo.
Tricky part is that now installation requires two steps - first installing firmware with partitions, and then uploading filesystem image.

MKloberg, 2023-10-27:
Hello Kirill,
 I'm the designer behind the Taurus Cooling Duct for the Creality E3 Sprite extruders and about to release V5, which includes a custom Knomi case that clips onto the front of the duct. Everything is pretty much ready on the mechanical and wiring side.
From here it's just customizing the Knomi and the release materials on Printables.
I came across your project as a fork and think that what you have accomplished so far is spectacular. Great work!!!
It's going in all the right directions, but there is no way that I can get my users (there may be +hundreds) to go through all the steps in VSCode to flash the thing. The use of littlefs makes it even more complicated.

What do you think about packaging it in this way? https://github.com/peakpeak-github/ImageMerge

The idea is that I design images/scenes that are more Taurus/Ender or even Creality branded to give this more traction.
I think I know how to do it and it would take more space, in which you come in because you already did the littlefs implementation.

For myself, I have successfully compiled your current build (not the release) and uploaded it to the Knomi.
It took it, but is blank when it starts up and doesn't respond on the BTT wifi. My guess is that it is missing the file system bin still, how do I upload that?

DiverOfDark, 2023-10-27:
Hi Mac,
and thanks for your nice words!

ImageMerge is certainly a nice possible way to package firmware update as part of release.

Regarding on how to install locally from VS Code - you need to run Upload Filesystem Image goal from PlatformIO (once your Knomi in OTA mode): 
[image]

DiverOfDark, 2023-10-27:
Just checked the source code and idea of ImageMerge - unfortunately it won't work as is because of out OTA update possiblity and partitions:
https://github.com/DiverOfDark/KNOMI/blob/master/Firmware/partition.csv

But basic idea to craft .bin file with app0+app1+zeroes as coredump+spiffs is a certain way to go. And distributing it with osptool.exe and .bat file to install also sounds like a nice idea.

DiverOfDark, 2023-10-27:
Mac, let's make a deal - if you'll make a PR with a single custom theme for Knomi (be it Taurus/Ender/Creality-based) then 
I'll implement a PR with single .bin (akin to ImageMerge) + a way to install with .bat and esptool.

DiverOfDark, 2023-10-29:
@MKloberg , please check release 1.2.0 - I've fixed an issue with Access Point not opening up + added single-file installer (rs-flasher) for firmware installation.

MKloberg, 2023-10-29:
@DiverOfDark I will, currently in a phase to put it all together mechanically. I will check out 1.2 tomorrow, but that will totally fix something that I had major point of entry problems with, hopefully making this easier for users.
Here's a recent rendering of the entire thing.
[image]

DiverOfDark, 2023-10-29:
Looks awesome!
Also, re theming - please note that due to hardware requirements it can't show gifs full-screen (240x240). 
Firmware needs to load image to ram, and we have about 80k free ram for image. It uses 3 bytes per color, so for e.g. image 150x150 it would require around 68k of ram. 
So in essence I would avoid even trying to load gifs larger than 150x150.

MKloberg, 2023-10-30:
Thank you! :-)
Sorry for my noobness, I didn't know the rover/platform was that limited. I can see it now in the partition table CSV, we only have 1MB to work with and even the crude BTT default images already take 930k of that up. Wow, that's very disappointing but I have stick to my release schedule and make it work somehow.
I'll flash with 1.2 now, see how that goes and want to change at least the Voron logo and the standby face to something else.
Thank you for working with me on this :-)

DiverOfDark, 2023-10-30:
Yeah, the KNOMI hardware is very limited, unfortunately. If only BTT soldered additional flash/ram on board - it could be way better and easier to use. Nevertheless we could try to make a custom prebuilt theme / binary with your images - as long as they are of reasonable size and fits inside fw ;)

I am also starting to think whether it could be worth to port this firmware to Mellow Halo as well - because it has better specs. Just compare:
Knomi - Flash 4Mb, Ram 320Kb
Halo - Flash 16Mb, Ram ?? (I suppose some external SRAM was soldered on board, but haven't found reliable specs yet).

I don't have Halo yet - so it is all speculations and hand waving so far :)

DiverOfDark, 2023-10-30:
Re GIFs limitation - raised #53, apparently there is a technical way to fix that, but that may require major firmware overhaul, so it's not something coming tomorrow.

MKloberg, 2023-10-31:
Hello Kirill,
 I tried rs-flasher and that didn't work at first, it found the port that the Knomi was on (COM4), but then gave me a connection error although I could see the CH340 port in device manager (nothing else was open at the time). Then I looked at the source code and noticed that the baud rate was much higher, so I forked your current build, built my own rs-flasher at 115200 baud (never done anything in rust, but figured it out), problem was the resources folder isn't in the project.
That worked and uploaded fine, but nothing would respond on IP 4.3.2.1 (captive didn't work on my iPhone). 
Running out of ideas, I downloaded the current release of rs-flasher again and then the upload suddenly worked with the current .exe. Still no captive, so I looked at the network info on my iPhone and found that it is now 192.9.200.1.

Summary: I documented this so you know, I'm not sure why it couldn't communicate with the port the first time around, the steps were exactly the same, but I have to say that rs-flasher is very, very cool. Thank you for that!!! No prerequisites, the image is inside of the executable. There is no better solution, that is as simple as it gets - even finding the port automatically, this is awesome. That the captive portal doesn't work is still an issue, not sure if that's fixable for all phones but we can always tell people to go to that IP address in the instructions or on screen.
Very nice, great progress! :-)

On the Halo idea, it's funny that you mention that. I have a few alpha printers involved on the Knomi case and one of them mentioned this to me. I didn't know it existed and that it is touch enabled too. I didn't know if you would even entertain switching to that/if that would even work, so I didn't say anything yet. So far it's only available on AliExpress and I ordered one this past weekend. Designing a new case for it will take some time and I'm running out of that commodity for the V5 release.

Early research on the Halo:
https://mellow.klipper.cn/?fbclid=IwAR2RFxt6__iPUPUT0jyWSUOjOsuTGGelHk0FM1-TRmbEeh9BFdcq9LquJVk#/board/fly_halo/
https://github.com/Mellow-3D/FLY-Halo?fbclid=IwAR34gzBP9kqbXTGkBhMgPSKodiwfQ1X2CI9_Ex-HN42jwDJkLxHzp74u1-Q

Until it arrives and I can play with it, I will stay on this path and do the images for the Knomi next, now that I know we have a way forward with rs-flasher that will certainly work for everyone it couldn't be simpler.

Overall, super cool - great work!!! 
Thanks,
Mac

closed 2024-05-12 by DiverOfDark