DiverOfDark/KNOMI #46: Optimise images at build

jMaddrell, 2023-10-29:
- Store high quality images in the repository
- Optimise GIFs at build time
- Allows for easier changes in the future

jMaddrell, 2023-10-29:
The metadata in each frame controls how long it is displayed for, the source images have duplicate frames to control animation times. They could potentially be removed by crafting the metadata for each frame. This would allow the AP_Config to be restored as well.

DiverOfDark, 2023-10-30:
Also for the newer / alternative images we need to optimize / crop each frame. In original images each frame have different size and different offset.
I was able to write some kind of Py script to repro that behavior, but not completed it fully - need to check pixel boundaries not against bg color, but against previous frame.
The thing is, pygif library provides really nice low-lever api for gif editing - so that you could, for example, reogranize frames/delays as you wish. Here's what I have crafted so far - https://gist.github.com/DiverOfDark/bdcb5fddb2499d465f2a312d2f03a585

jMaddrell, 2023-10-30:
I've tried to use imagemagick to remove the duplicate frames but they are different enough that it doesn't remove anything. Will take a look into the custom script approach.

jMaddrell, 2023-11-04:
It may be worth removing the black background from the stock images, this would allow users to change the background colour themselves.

DiverOfDark, 2023-11-04:
2 concerns about transparent gifs:
 - I would check performance of rendering of transparent gifs before actually implementing that.
 - This would only solve issue for background color, but we now have accent color as well. And I doubt we'll find a way to apply it to gifs (I would prefer to avoid tinting/color shifting image at runtime). So may be it's just not worth of effort.

closed 2025-11-05 by jMaddrell