DiverOfDark/KNOMI #51: Idea: custom append-only pseudo-readonly-fs + ability to defrag it instead of littlefs

DiverOfDark, 2023-10-30:
Problem statement:
Currently rendering a BMP file from littlefs takes enormous amount of time. It also leads to memcpy and byte copying.

Idea:
We could replace littlefs with custom binary blob with flat structure (files as contiguous blocks) and replace using of filesystem to getting a pointer to file in filesystem. e.g. structure:

- Header
- Files count (N)
- [1..N] file metadata
- -  name
- -  offset
- -  isDeleted
- [1..N] raw file contents
- FreeSpace

Pros:
- access to file contents almost instantly - just pointer to memory region
- could solve #13 
- could reduce memory allocations along the way

Cons:
- complex
- not some industry standard - so a custom code for integrating it + custom code to build filesystem image.
- For use-case of uploading file - we need to delete old file. This leads to fragmentation. We could write code to defragment FS after file upload. This is a tricky and risky operation, suspecible to fails on power loss - so must be done carefully. Also it can take significant amount of time. 
- Theme-related settings file, e.g. accent color. If we store them in JSON - this would take significant amount of time to actually update them.

@jMaddrell , what do you think?

jMaddrell, 2023-10-30:
I would personally avoid implementing a custom FS even if there are some benefits, avoiding things like corruption in the event of powerless during write would be my biggest concern. 

To work around that we would likely need a metadata block that contains the current pointer to each fs entry, and use COW/copy-on-write semantics. Which would quickly cause issues given how little storage these devices have available. 
We would also have to come up with a way of defragmenting the storage which would likely become complex.

However without being able to use mmap() to map the fs into memory I can't see how this would be quicker than littlefs/fatfs?

DiverOfDark, 2023-10-30:
If all files are contagious - meaning not stored in chunks, but as one block - then as this is same storage as general code storage - meaning it is already mapped to memory.

DiverOfDark, 2023-10-30:
Anyway - I would focus on #53 instead of custom fs and see if it fixes most of the problems

jMaddrell, 2023-10-30:
Sounds good to me

closed 2023-11-01 by DiverOfDark