DiverOfDark/KNOMI #55: The printer status is not transmitted.

Yackushev, 2023-10-31:
I flashed Knomi using rs-flasher on version 1.2.0 and 1.2.1.
The settings window is working.
If you update knomi through the settings window, the completion percentages are displayed.
But knomi does not transmit the printer status. Shows eyes and Voron logo.
No printing percentage, no heating - nothing.
If you specify the wrong printer address, knomi writes that there is no connection.
I tried it on two printers - it doesn't work the same.
Perhaps I missed something, although I checked many times.
Thank you.

DiverOfDark, 2023-11-01:
Can you share a bit details:
1) do you have bed heating?
2) you actually started printing? (not just pre-heat)
3) if you open Knomi page and hit F12 to see dev console - there's stream of logs from Knomi.

Can you dump them for some time during printing (literally 10 sec would be enough) and attach to this issue?

Yackushev, 2023-11-01:
Yes, I tried turning on the heating of the table and nozzles.
Connected while printing.
Nothing happened.
I'm not sure if I saved the right data, but here's what I have.
[image]

DiverOfDark, 2023-11-01:
On the bottom of screenshot you have "Console" section with logs - that is what required.
Could you please:
1) start printing something (not just heating up) 
2) open this console, so that it starts getting logs
3) wait for 1 minute
4) copy-paste all messages from "Console" and attach here.

DiverOfDark, 2023-11-01:
Yeah, that was it. Will try to dig into it and/or increase logging to get better details on what happens.

Yackushev, 2023-11-03:
Could you post the compiled firmware with the latest changes for rs-flasher?
Unfortunately, I can't compile it myself. I'm getting too many errors.

DiverOfDark, 2023-11-03:
Will do once properly test the PR and merge it.

DiverOfDark, 2023-11-04:
Should be fixed in 1.3.0, please check.

closed 2023-11-04 by DiverOfDark

Yackushev, 2023-11-05:
Yes, everything works. Thank you :)
I would like to know if it is possible to remove the voron logo and leave only one animation in standby mode?

DiverOfDark, 2023-11-05:
You could use webui and upload standby animation as Voron.gif

Yackushev, 2023-11-05:
Yes, I've already tried that.
But you can see a jump when the animation changes, and it looks even worse.

DiverOfDark, 2023-11-05:
Raised #62 to alleviate this effect of jumping. No other solution out of the box yet.