DiverOfDark/KNOMI #59: 1.2.1 Scenes Not Switching

MKloberg, 2023-11-02:
[Creality Ender3 S1 Pro on Sonic Pad]
First time trying to print with Knomi on the DOD/Knomi firmware and littlefs using 1.2 rs-flasher.
Flashing went great, it took it - I then did the wiring yesterday to grab 5V off the CR Touch probe, that is a stable power source.
I had that working in OTA mode as well and hit a few of the URL's to see what's on the file system, etc. All good.
While printing or even warming up, only the homing and leveling scenes are displaying. It remains in standby (Voron.gif and the face) any other time (even during warm up and printing).
Only when the Klipper Macros run for homing and leveling, the display changes.
Occasional reboots also, but they are so fast - hardly noticeable.
I checked all calls against the Klipper API, that worked fine from a different device so that's not the problem.

I compiled a 1.2.1 debug image off the current fork and loaded that. Same thing.
Included is the monitor log, perhaps you can tell what is happening.
(i blanked out my WiFi password in the log, the only change)

[Knomi-1.2.1-Log-mkloberg.log](https://github.com/DiverOfDark/KNOMI/files/13233819/Knomi-1.2.1-Log-mkloberg.log)

DiverOfDark, 2023-11-02:
Looks like the same as #55 , thank you, looking into it.

MKloberg, 2023-11-02:
I built new images from the branch you recommended and re-flashed. It's a little better now, it displays the heating scenes which it didn't do before, the homing scene as before and the ABL scene as before, but when it comes to the rest, if falls back to standby (voron and the face) while the printer is actually printing. I checked this before, but here's what the Klipper status api said while it was displaying standby during the print.

Attached is a full log file of the print, start to end.

{
    "temperature": {
        "bed": {
            "actual": 75.15,
            "offset": 0,
            "target": 75
        },
        "tool0": {
            "actual": 246.81,
            "offset": 0,
            "target": 245
        }
    },
    "state": {
        "text": "Printing",
        "flags": {
            "operational": true,
            "paused": false,
            "printing": true,
            "cancelling": false,
            "pausing": false,
            "error": false,
            "ready": false,
            "closedOrError": false
        }
    }
}

[logfile2.txt](https://github.com/DiverOfDark/KNOMI/files/13245667/logfile2.txt)

I also still noticed occasional reboots during printing, where do you think this is coming from?
Looks network related to me.

[INFO]  [109840]        (200.650, +11)   processJson: 75°C      (in src/network/requests/Request1.h line #27)
[INFO]  [107984]        (200.754, +104)  processJson: false     (in src/network/requests/Request3.h line #9)
[INFO]  [107828]        (200.757, +3)    processJson: false     (in src/network/requests/Request4.h line #11)
[INFO]  [107676]        (200.771, +14)   processJson: 60        (in src/network/requests/Request2.h line #18)
[INFO]  [108628]        (202.341, +1570)         Execute: Executing request, state is 4 (in src/network/requests/KlipperApiRequest.h line #41)
[INFO]  [107912]        (202.353, +12)   Execute: Executing request, state is 4 (in src/network/requests/KlipperApiRequest.h line #41)
[INFO]  [107172]        (202.354, +1)    Execute: Executing request, state is 4 (in src/network/requests/KlipperApiRequest.h line #41)
[INFO]  [106444]        (202.366, +12)   Execute: Executing request, state is 4 (in src/network/requests/KlipperApiRequest.h line #41)

assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

Backtrace: 0x400835c5:0x3ffd4420 0x4008b945:0x3ffd4440 0x40090f11:0x3ffd4460 0x401190de:0x3ffd4590 0x4011918c:0x3ffd45b0 0x400eb602:0x3ffd45d0 0x40115f80:0x3ffd45f0

  #0  0x400835c5:0x3ffd4420 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008b945:0x3ffd4440 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40090f11:0x3ffd4460 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x401190de:0x3ffd4590 in tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:951
      (inlined by) tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:931
  #4  0x4011918c:0x3ffd45b0 in tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:991
  #5  0x400eb602:0x3ffd45d0 in _tcp_recved_api(tcpip_api_call_data*) at .pio/libdeps/debug/AsyncTCP-esphome/src/AsyncTCP.cpp:444
  #6  0x40115f80:0x3ffd45f0 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:172
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: a2d81259e5655a68

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[INFO]  [215904]        (0.158, +158)    setup: Setup   (in src/main.cpp line #52)
[INFO]  [213972]        (0.165, +7)      setup: LittleFS started        (in src/main.cpp line #54)
[   170][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
[INFO]  [213672]        (0.170, +5)      setup: Config created  (in src/main.cpp line #56)
[INFO]  [213636]        (0.181, +11)     setup: WifiManager created     (in src/main.cpp line #58)
[INFO]  [213544]        (0.392, +211)    setup: Timer and button created        (in src/main.cpp line #60)
[INFO]  [212492]        (0.746, +354)    setup: DisplayHAL created      (in src/main.cpp line #62)
[INFO]  [211344]        (0.747, +1)      setup: KlipperAPI started      (in src/main.cpp line #64)
[INFO]  [208484]        (0.759, +12)     KnomiWebServer: WebServer started!     (in src/network/KnomiWebServer.cpp line #329)
[INFO]  [208484]        (0.769, +10)     setup: WebServer started       (in src/main.cpp line #66)
[INFO]  [208380]        (0.840, +71)     BootupLogoScene: Loading boot logo     (in src/scenes/BootupLogo.h line #16)
[INFO]  [179100]        (0.870, +30)     ResourceImage: Created resource image /BTT_LOGO.gif    (in src/ui/ResourceImage.h line #37)
[INFO]  [179164]        (0.870, +0)      BootupLogoScene: Boot logo loaded      (in src/scenes/BootupLogo.h line #18)
[... cut ...]
[INFO]  [132912]        (1.062, +124)    WifiStation: Connecting to WIFI MySpectrumWiFi10-5G / stealthtree476   (in src/network/WifiStation.h line #13)
[INFO]  [131808]        (2.100, +1038)   connectToWiFi: STA connected   (in src/network/WifiManager.h line #67)
[INFO]  [131808]        (2.100, +0)      connectToWiFi: WIFI connect Success    (in src/network/WifiManager.h line #71)
[INFO]  [131760]        (2.111, +11)     connectToWiFi: SSID:MySpectrumWiFi10-5G        (in src/network/WifiManager.h line #72)

MKloberg, 2023-11-02:
I will say that my startup sequence is optimized.
- Heat the bed and the extruder to standby
- Homing
- ABL
- Move to start purge
- Heat extruder to final
- Print

So there an extra heating step after ABL, does that matter or could the sequence confuse the Knomi logic?

DiverOfDark, 2023-11-03:
Ok, found a culprit - and pushed changes to aforementioned branch. Please try to install again and verify.

Root cause: when deserializing json we have to specify buffer size which should fit all json data. because buffer size was too small - "state" part of response was missing. Also added to log a parsed json message to double check that it is the same as received from printer.

DiverOfDark, 2023-11-03:
Re occasional reboot - see #10.

Bazeone, 2023-11-08:
Still not switching screens for homing or probing. Bed temp and Hotend temp work.

DiverOfDark, 2023-11-09:
Did you do changes in klipper config, like original guide suggested?

MKloberg, 2023-11-10:
Currently too much to do with the release of Taurus V5 and running the original firmware, which works.
I can say that when I ran into this, everything klipper config was in place like it is now, BTT firmware works without any reboots or skipping scenes on my rig.

Another thing I can say is that the whole time, the BTT logo and the way that circle was going around after it boots was not there with this firmware, it hung at the first segment (about the top right), stopped there and it would take a while until the Voron logo appeared. The logo itself looked not as smooth as in the stock firmware. I'm not sure why all that is, but wanted to point it out.

[image]

Bazeone, 2023-11-11:
> Did you do changes in klipper config, like original guide suggested?

I had it working before flashing this firmware.

DiverOfDark, 2023-11-11:
Reopened this issue because of homing/bed leveling not displayed.
Overall stability and smoothness will be fixed under #10 .

MKloberg, 2023-11-11:
Knomi V2 came out a few days ago which seems to be OTA out of the box, is touch enabled and has a camera interface. I will look into that for my project, I'm supposed to get one tomorrow. I haven't seen the repo for it yet, we'll see.

commit 33436655 (2023-11-03): Resolves #59, #55
PR #152 (2024-02-17): Feature/homing ang qgling
Added newer gifs
Fixing homing / adding qgling

Resolves #127 
Resolves #59 
Resolves #140
closed 2024-02-17 by jMaddrell