DiverOfDark/KNOMI #127: bed mesh gif and wifi

hendr13x, 2024-01-19:
firstly well done on all of this its damn impressive and atm well above my head the work involved.

is it possible to implement a gif for bed meshing?

and is their a way to either add a delay into when it connects to the wifi on start up or for it to keep tying to connect till it does?

my issue wwith the wifi i beleve is due to the fact my voron has its own wifi built into it and it takes it longer to boot than the knomi does so i have to connect to knomi wifi each time i turn on the printer to save the wifi info to get it to connect.

Thanks for all the work you have into this.

jMaddrell, 2024-01-29:
There is currently a GIF for leveling, that should show.

jMaddrell, 2024-01-29:
In terms of changing the WiFi connect timeout I think you would just need to increase https://github.com/DiverOfDark/KNOMI/blob/7280bd8e71bfa2220da5ca31b7cb879ca4d8feab/src/network/WifiManager.h#L39 

How long does it usually take your machine to boot?

hendr13x, 2024-02-10:
Hi mate, my printer takes bout 60 seconds to boot a 65 - 70 second timeout would be great. 
is that something you could impliment easily? 
i will have a look into it but have little to no idea on how to make the firmware.

thank you

DiverOfDark, 2024-02-17:
QGL is in PR, will be in next version.
WiFi duplicates #97

closed 2024-02-17 by jMaddrell

DiverOfDark, 2024-02-17:
QGL - Try use 2.0.0 (update both FW and Theme/Files) + include knomi.cfg from repository to your klipper config

PR #152 (2024-02-17): Feature/homing ang qgling
Added newer gifs
Fixing homing / adding qgling

Resolves #127 
Resolves #59 
Resolves #140