DiverOfDark/KNOMI #239: Bricked KNOMI v2

mkryuchkov, 2024-05-14:
Hello!

Flashed my KNOMI v2.2.0 with rs-flasher. What I have for now:

- web interface and settings (wifi, hostname, etc.) working
- KNOMI is showing Voron logo and idle face animations one after another in infinite loop (is it boot/crash loop?)
- klipper config brake macroses, but managed to send homing status
- KNOMI is showing homing animation
- KNOMI has never responded to touches and I haven't see menu at least once
- KNOMI can enter flash mode and PC sees COM device
- Can't flash with ESP32 flash tool having `COM FAIL` error not btt FW nor 2.2.0 full one.
- Can't flash with rs-flasher having `Firmware flashing error! // Error while connecting to device`

KNOMI seems to be broken.
What happened? What possible solutions are?

DiverOfDark, 2024-05-15:
Hi!

TLDR; - this fork is originally based on V1 firmware (even for V2) - meaning it doesn't support menu / touchscreen (yet, maybe?), and it is mostly rewritten from scratch - so it's not an easily interchangeable with original one.

>>KNOMI is showing Voron logo and idle face animations one after another in infinite loop (is it boot/crash loop?)
It is not, it is expected behavior from Knomi V1

>>klipper config brake macroses, but managed to send homing status
Could you elaborate here a bit - what exactly did broken and what did you change? (this also might be the reason why you get strange behavior later)

If you want to revert to original BTT firmware - I'll try on my device and will try to write a guide on how to flash original fw.  
Good news is - you can't brick device alltogether - esp32(s3) are really errorprone, but also they are not very friendly for non-developers. Probably you'll have to use esptool to flash partitions and fw from original fw.

mkryuchkov, 2024-05-16:
Managed to build and flash stock firmware

mkryuchkov, 2024-05-16:
@DiverOfDark 
Thank you for reply!

Yeah, I didn't see it implements V1 features only, my bet.

Macroses. I figured out these are just examples for override. Since I use my own overrides I have to adopt them adding knomi commands in there. I just tried to copy-paste as manual intended.

It would be nice to have a guide or some predefined options like menu-option in rs-flasher you made.
Can help here, as I managed to do that.

Yeah, S3 is more user friendly than I expected. Can be software enabled to flash mode and hard to kill by incorrect FW. Saved me :)

mkryuchkov, 2024-05-16:
How to revert to stock.

- Get bootloader, partitions and firmware bin files from official repo ([v1](https://github.com/bigtreetech/KNOMI/tree/master/KNOMI1/Firmware), [v2](https://github.com/bigtreetech/KNOMI/tree/master/KNOMI2/Firmware)). Or you can build it yourself from sources.
- Get [proper usb driver for windows](https://www.wch-ic.com/downloads/CH341SER_EXE.html) (was necessary in my case). Your device should be okay from OS point: 
[image]
COM device without any errors.

- Get flash tool of choice
  - esptool (CLI, universal option)
  - flash download tool (windows only, mentioned in [V1 btt guide](https://bttwiki.com/KNOMI.html#update-firmware))
- Flash firmware
  - For V1 you may use [official guide](https://bttwiki.com/KNOMI.html#update-firmware) as is
  - For V2 you may use the same **BUT**
    - Choose **esp32s3** instead of esp32 starting the flash download tool
    - Use **0x0000** offset for bootloader instead of 0x1000
[image]

  - Both could be flashed with esptool [write flash](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#write-binary-data-to-flash-write-flash) command

Build firmware is not necessary since you may use prebuilt ones from BTT repo or this repo. In case you are interested in just google PlatformIO + VSCode + ESP32, there are some good videos on YT.

DiverOfDark, 2024-05-18:
I'd like rs-flasher to be able to restore original BTT firmware - so will reopen this ticket for that feature.

DiverOfDark, 2024-05-19:
Will be in next release as part of rs-flasher.
Related PR - https://github.com/DiverOfDark/KNOMI/pull/242.

closed 2024-05-19 by DiverOfDark