DiverOfDark/KNOMI #241: The Newest Firmware (v2.2.0) Not Working With Knomi-V1

dhierholzer, 2024-05-16:
Hello,

I attempted to update my Knomi-V1 firmware from V2.1.1 (Your Firmware), to V2.2.0 without sucesss.

What I Tried:
First, I tried just pushing the update using the browser and uploading the OTA version of the firmware. Once completed, this just caused my knomi-V1 device to start "flickering" a blank backlit white screen on & off. 
When I say flickering, it was more like on for 2 seconds, off for 2 seconds* in a continuous loop.
Note: I attempted this when my Knomi-V1 device was being powered via the EBB RP2040 board; connected to my Stealthburner Extruder.

With This not yielding any fruitful results, I then extracted my Knomi-V1 device from my Stealthburner, and connected it directly to my Desktop PC via USB C.
Since It was now connected to my Desktop PC, I attempted to use the newest rs-flasher executable to see if it would provide better results.
The rs-flasher ran without any issues and once completed, told me to unplug my device and plug it back in as per usual.
Once I plugged my device back in, I unfortunately got the same result as I did previously with the OTA firmware update.

So once again, my device was just a blank backlit white screen that turned on and off every 2 seconds*.

What I Did Next:
Because I was not having any luck with this newest Firmware Update, I went ahead and re-flashed my device with the previous firmware version (V2.1.1), and thankfully my device is working again.

Thanks For all of your work creating this new custom firmware for these ESP32 Devices!

Cheers!

_** 2 seconds was not timed and instead is just an estimation_

DiverOfDark, 2024-05-17:
Thanks for raising that - I've updated the release tag to clearly show that it doesn't work on Knomi V1.
Will have to disassemble my Stealthburner to test - so might take some time.

DiverOfDark, 2024-05-19:
Will be in next release as part of rs-flasher.
Related PR - https://github.com/DiverOfDark/KNOMI/pull/242.

closed 2024-05-20 by DiverOfDark

DiverOfDark, 2024-05-20:
Fixed in 2.2.1 - please check, if you can.

PR #249 (2024-05-20): Fix knomi V1 
Resolves #241.