DiverOfDark/KNOMI #298: only shows % done

KennyVB, 2024-07-25:
after updating to this version. everything works. but when printing. it only shows % printing done
what giuves?=

DiverOfDark, 2024-07-27:
Could you attach the log during printing?

You can obtain it by opening Knomi webpage, and opening developer tools (F12), e.g.:

[image]

KennyVB, 2024-07-28:
@DiverOfDark here you go. still printing tho
[10.0.2.185-1722202406845.log](https://github.com/user-attachments/files/16405286/10.0.2.185-1722202406845.log)

DiverOfDark, 2024-07-29:
The only strange thing I see in the log - is unexpected frame received from klipper - so pushed a v2.3.2 version.

Also I should have asked to elaborate what exactly the problem is.
I initially had an assumption that display shows just "%" sign instead of progress during printing.

Could you share what you expectations are (like what is happening, and what you'd expect to happen) and which device do you have?

If you have a Knomi V2 and expect touchscreen to react - then unfortunately it's not yet implemented. (See README.MD)
Also if you want touchscreen menus - could you share what exactly you want to control through it, like which use-cases?
(because Im personally struggle to see much value in it, and normally just use web ui to control printer)

KennyVB, 2024-07-31:
hey
i got a Knomi v1, so all good there.
well, in the "theme" it has a lot of gifs
standy i ojnly see the eyes looking side to side then logo back and forth
home, gantry and bed mesh it shows the gifs
but printing is only % done. never the printing face it has, swirvly eyes etc. 
would love it to change between temp, priting gif, and ofc % done. that's how i think it should or was intended. boring to look at % only and rare glimps of the other faces for 1 milli secound when starting print and bam % done only. i'll atacch a photo next print, of what it doest fully

KennyVB, 2024-08-11:
ohh you set it as enchancement.. i just thought it was standard that it did stuff. and not just % done when printing. but changed between it all

well new issues raise :)
klicky probe = knomi 
now g28, quad gantry lvl dont work :P